altera2015/usbserial #2: the listDevices always null

zxj17815, 2019-03-05:
I simulation usb COM1(android)  and usb COM2,but the listDevices always null

altera2015, 2019-03-05:
I'm guessing your solution is not simulating a USB CDC (or similar) device but directly a COM port in the Linux kernel. This code base is intended to access UART hardware connected via USB.
Can you give me more details on how you simulated your comports? What hardware vendor and product ids?

Thanks

zxj17815, 2019-04-22:
I can connect now, but the problem is that I did not receive the returned information after sending, I do not know whether the instruction sent in the past does not belong to hex? Because I can receive the return message when I test with another software

zxj17815, 2019-04-23:
I find the  pid=24577&vid=1027，it doesn't work;
and I use  the other pid=8963&vid=1659, it work.

zxj17815, 2019-04-23:
oh,but I tested by FTDIs'app ,it's working, I have no idea...

zxj17815, 2019-04-26:
Is it a driver problem? But open is ok, when I send the data, the receiving light on the control panel flashes, does it mean that the hardware and driver are ok, just my code has problems.

altera2015, 2019-05-04:
I'm not sure, can you link your code?

altera2015, 2019-05-09:
I've updated to the latest UsbSerial underlying library. Give it a go with 0.2.1, it might solve your trouble.

zxj17815, 2019-05-14:
Thank you, it work now

closed 2019-05-14 by zxj17815