altera2015/usbserial #4: Add support for parity even

Salabasti, 2019-05-07:
In usb_serial version 0.1.1 there seems to be no constant for setting the parity of a UsbPort object to 'even'.

I explored the underlying [java library](https://github.com/felHR85/UsbSerial) and noticed that this option is supported there. ([link](https://github.com/felHR85/UsbSerial/blob/3465c5041e2ec4258c3b705d35ead60435d15592/usbserial/src/main/java/com/felhr/usbserial/UsbSerialInterface.java#L22)) I tried using '2' as a constant for setting the parity in UsbPort.setPortParameters() with no success. 

Will the option to set the parity to 'even' be available in the near future or is there any workaround for that?

altera2015, 2019-05-07:
I've added the 'Even' constant in 0.2.0

Salabasti, 2019-05-08:
Using an Integer value of 2 seems also to work for setting the parity argument to 'even'.

Salabasti, 2019-05-08:
> I've added the 'Even' constant in 0.2.0

Thank you very much!

closed 2019-05-08 by Salabasti