altera2015/usbserial #7: The library does not support ABACUS ELECTRICS Optical Probe

vnlevanduoc, 2019-06-24:
Hello guys!

I am learning about USB serial in Flutter language and see that your library is supporting this issue

But I import the library and run the example code, I get an error like this

When running to the section: "_port = await device.create ();" then it reports an error
"[ERROR: flutter / lib / ui / ui_dart_state.cc (148)] Unhandled Exception: PlatformException (UsbSerialPortAdapter, Not an Serial device., Null)"

Can you support me?

p / s: please I am not fluent in English so thanks to google translation, I hope you guys will apologize

THANK YOU!

vnlevanduoc, 2019-06-25:
I tried adding one parameter "ftdi" to the "device.create ()" function, the program reported the connection

But I want to send and receive back as a byte, what do I use?

altera2015, 2019-07-18:
This library is higher level than the bulkTransfer call. UsbSerial provides a wrapper layer exposing the Serial USB devices, not the raw devices.

closed 2019-07-23 by altera2015