altera2015/usbserial #9: java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread.

SamuelAG, 2019-07-10:
I tried to use your example but it is giving a exception:

`E/AndroidRuntime(30424): FATAL EXCEPTION: Thread-5
E/AndroidRuntime(30424): Process: dev.bessems.usbserialexample, PID: 30424
E/AndroidRuntime(30424): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: Thread-5
E/AndroidRuntime(30424): 	at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:794)
E/AndroidRuntime(30424): 	at io.flutter.embedding.engine.FlutterJNI.dispatchPlatformMessage(FlutterJNI.java:684)
E/AndroidRuntime(30424): 	at io.flutter.embedding.engine.dart.DartMessenger.send(DartMessenger.java:80)
E/AndroidRuntime(30424): 	at io.flutter.embedding.engine.dart.DartExecutor.send(DartExecutor.java:174)
E/AndroidRuntime(30424): 	at io.flutter.view.FlutterNativeView.send(FlutterNativeView.java:144)
E/AndroidRuntime(30424): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:226)
E/AndroidRuntime(30424): 	at dev.bessems.usbserial.UsbSerialPortAdapter$1.onReceivedData(UsbSerialPortAdapter.java:60)
E/AndroidRuntime(30424): 	at com.felhr.usbserial.UsbSerialDevice$WorkerThread.onReceivedData(UsbSerialDevice.java:361)
E/AndroidRuntime(30424): 	at com.felhr.usbserial.UsbSerialDevice$WorkerThread.doRun(UsbSerialDevice.java:336)
E/AndroidRuntime(30424): 	at com.felhr.usbserial.AbstractWorkerThread.run(AbstractWorkerThread.java:21)`

When I remove this part of the code, the error disappears
`_subscription = _transaction.stream.listen((String line) {
      setState(() {
        _serialData.add(Text(line));
        if (_serialData.length > 20) {
          _serialData.removeAt(0);
        }
      });
    });`
But i need to receive data from UART.

Flutter Doctor:
`Doctor summary (to see all details, run flutter doctor -v):                                                             [√] Flutter (Channel stable, v1.7.8+hotfix.3, on Microsoft Windows [versÃ£o 10.0.17134.829], locale pt-BR)                                                                                                                                      [√] Android toolchain - develop for Android devices (Android SDK version 28.0.3)                                        [!] Android Studio (version 3.4)                                                                                            X Flutter plugin not installed; this adds Flutter specific functionality.                                               X Dart plugin not installed; this adds Dart specific functionality.                                                 [√] VS Code (version 1.36.0)                                                                                            [√] Connected device (2 available) `

Any ideia what is might be?

Eternali, 2019-07-16:
I had this issue too, it's been cropping up in a lot of plugins recently. I think it has to do with plugins needing to communicate with dart on the main thread rather than some background thread. The workaround to fix this is basically to edit `UsbSerialPortAdapter.java` to replace this:
```
if ( m_EventSink != null ) {
    m_EventSink.success(arg0);
}
```
with this:
```
if ( m_EventSink != null ) {
    m_handler.post(new Runnable() {
        @Override
        public void run() {
            m_EventSink.success(arg0);
        }
    });
}
```
Where `m_handler` is set in the constructor to be `m_handler = new Handler(Looper.getMainLooper());`.

Note: you have to add these imports too
```
import android.os.Handler;
import android.os.Looper;
```
I'll have a PR up for this shortly.
Hope it helps!

altera2015, 2019-07-21:
Thanks for the PR, it's pulled and published!

PR #10 (2019-07-18): fixed uithread error when subscribing to usb device
closes #9