altera2015/usbserial #12: Ask about _transaction.stream.listen

vnlevanduoc, 2019-08-09:
I use this example: altera2015 / usbserial / blob / master / example / lib / main.dart
see after "await _port.write (Uint8List.fromList (data.codeUnits));" all the results returned will be listened through the function "_transaction.stream.listen ((String line)" and print the result.

So it is similar to the _port.read () function of the corner library (https://github.com/felHR85/UsbSerial)?

vnlevanduoc, 2019-08-10:
I have the python code running normally
`
def printLine(message, dir):
	if type(message) != "str":
		message = str(message)
	m2 = ""
	for m in message:
		c = m #binascii.hexlify(m)
		if c == "01":
			m2 += '<SOH>'
		elif c == "02":
			m2 += '<STX>'
		elif c == "03":
			m2 += '<ETX>'
		elif c == "0a":
			m2 += '<LF>'
		elif c == "0d":
			m2 += '<CR>'
		elif c == "06":
			m2 += '<ACK>'
		elif c == "15":
			m2 += '<NAK>'
		else:
			m2 += m

	print(dir + " " + m2)

	if dir == "RX":
		print("")

def getLine():
	res = []
	while True:
		c = s.read()
		if not c: break
		res.append(c)
	res = b''.join(res)
	print('{:%Y-%m-%d %H:%M:%S}'.format(datetime.datetime.now()) + '   <<<   : ' + res.decode('utf-8'))
	return res

def sendLine(message):
	s.write(message)
	printLine(message, 'TX')

def getId():
	message = bytearray(b'\x2F\x3F\x21\x0D\x0A')    #Sends command: /?!<CR><LF> to begin communication
	sendLine(message)
	return getLine()

def modeSwitchRequest():
	message = bytearray(b'\x06\x30\x35\x31\x0D\x0A')
	s.write(message)
	time.sleep(0.2)     #Waits to send the whole message before changing port setup.
	s.baudrate = 9600
	return getLine()
`
but transfer to your library

`Future<String> _sendMsg(Uint8List msg) async {
    int flag = 1;
    String response;
    while (flag < 5) {
      response = await _transaction.transaction(_port, msg, Duration(seconds: 5));
      if (response == null) {
        flag = flag + 1;
      } else {
        flag = 5;
      }
    }
    return response;
  }

  Future<String> _getId() async {
    var response = await _sendMsg(Uint8List.fromList([0x2F,0x3F,0x21,0x0D,0x0A]));
    return response.toString();
  }

  Future<String> _modeSwitchRequest() async {
    _port.write(Uint8List.fromList([0x06,0x30,0x35,0x31,0x0D,0x0A]));
    sleep(const Duration(microseconds: 200));
    _port.setPortParameters(9600, UsbPort.DATABITS_7, UsbPort.STOPBITS_1, UsbPort.PARITY_EVEN);
    //???????? => I don't understand

    
    // var response = await _sendMsg(Uint8List.fromList([0x06,0x30,0x35,0x31,0x0D,0x0A]));
    // if(response != null){
    //   _port.setPortParameters(9600, UsbPort.DATABITS_7, UsbPort.STOPBITS_1, UsbPort.PARITY_EVEN); //Set lại baudrate
    //   return response;
    // }
    return "";
  }`

for the "getId ()" function we print send and print normal results

But when using the "modeSwitchRequest ()" function, it returns null, I don't understand the reason

p/I tried using normal transactions and write still not returning results

vnlevanduoc, 2019-08-17:
I have a python code like this
```import serial
import time
import binascii
import datetime

s = None
password = "FEDC0003"         #default password level 2 ABCD0002

def open():
	global s

	s = serial.Serial()
	s.port = "COM4"
	s.baudrate = 300
	s.bytesize = serial.SEVENBITS           #number of bits per bytes
	s.parity = serial.PARITY_EVEN           #set parity check: EVEN
	s.stopbits = serial.STOPBITS_ONE        #number of stop bits
	s.timeout = 1.5                         #non-block read
	s.xonxoff = False                       #disable software flow control
	s.rtscts = False                        #disable hardware (RTS/CTS)
	try:
		s.open()
		print ('{:%Y-%m-%d %H:%M:%S}'.format(datetime.datetime.now()) + '   ###   :' + 'Begining communication')
	except Exception as e:
		print ("Error attemping to open serial port: " + str(e))
		exit()

def printLine(message, dir):
	if type(message) != "str":
		message = str(message)
	m2 = ""
	for m in message:
		c = m #binascii.hexlify(m)
		if c == "01":
			m2 += '<SOH>'
		elif c == "02":
			m2 += '<STX>'
		elif c == "03":
			m2 += '<ETX>'
		elif c == "0a":
			m2 += '<LF>'
		elif c == "0d":
			m2 += '<CR>'
		elif c == "06":
			m2 += '<ACK>'
		elif c == "15":
			m2 += '<NAK>'
		else:
			m2 += m

	print(dir + " " + m2)

	if dir == "RX":
		print("")

def getLine():
	res = []
	while True:
		c = s.read()
		if not c: break
		res.append(c)
	res = b''.join(res)
	print('{:%Y-%m-%d %H:%M:%S}'.format(datetime.datetime.now()) + '   <<<   : ' + res.decode('utf-8'))
	return res

def sendLine(message):
	s.write(message)
	printLine(message, 'TX')

def getId():
	message = bytearray(b'\x2F\x3F\x21\x0D\x0A')    #Sends command: /?!<CR><LF> to begin communication
	sendLine(message)
	return getLine()

def HangUp():
	message = bytearray(b'\x01\x42\x30\x03\x71')    #Sends command: /?!<CR><LF> to begin communication
	sendLine(message)
	return getLine()

def modeSwitchRequest():
	message = bytearray(b'\x06\x30\x35\x31\x0D\x0A')
	s.write(message)
	time.sleep(0.2)     #Waits to send the whole message before changing port setup.
	s.baudrate = 9600
	return getLine()

def decryptA1700(seed, password):
	crypted = bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00')
	password = bytearray(password, 'ascii')

	for i in range (0, 8):
		crypted[i]= password[i] ^ seed[i]
	last = crypted[7]
	for i in range (0, 8):
		crypted[i] = (crypted[i] + last) & 0xFF
		last = crypted[i]

	crypted = (binascii.hexlify(crypted)).swapcase()
	return crypted

##########################################################	
open()

identidad = getId()

if identidad:
	seed = modeSwitchRequest()

	_str = 	str(seed[5:21])
	seed = bytearray.fromhex(seed[5:21].decode())

	pw = decryptA1700(seed, password)

	#	Send hashed password
	message = bytearray(b'\x50\x32\x02\x28')
	message.extend(pw)
	message.extend(b'\x29\x03')

	send = bytearray(b'\x01')
	send.extend(message)
	send.append(getCRC(message))

	sendLine(send)
	response = getLine()

	if response != b"\x06":
		print("Invalid response")
		HangUp()

	#	Request reading
	#message = bytearray(b'\x01\x52\x31\x02\x35\x30\x37\x30\x30\x31\x28\x34\x30\x29\x03\x64')
	message = bytearray(b'\x01\x52\x31\x02\x36\x38\x30\x30\x30\x31\x28\x31\x36\x29\x03\x6B')

	sendLine(message)
	response = getLine()

	message = bytearray(b'\x01\x52\x31\x02\x35\x30\x37\x30\x30\x31\x28\x34\x30\x29\x03\x64')
	sendLine(message)
	response = getLine()

	# print("Import:" + mImport)
	print(response)

HangUp()

s.close()```

###########end############################################################
Notice how this modeSwitchRequest function has reset the baudrate value

I have moved the above python code to the dart but to the _modeSwitchRequest function it returns nothing

I adjust the _modeSwitchRequest function again as follows

```  Future<String> _modeSwitchRequest() async {
    _port.write(Uint8List.fromList([0x06,0x30,0x35,0x31,0x0D,0x0A]));
    _port.write(Uint8List.fromList([0x0A]));
    await _port.setPortParameters(9600, UsbPort.DATABITS_7, UsbPort.STOPBITS_1, UsbPort.PARITY_EVEN);
    return "";
  }```

Can you separate the parameters in setPortParameters into multiple functions?

because it depends on the needs and timing, I do not always put it like that

altera2015, 2020-03-07:
Is this still an issue you need?

closed 2021-07-11 by altera2015