altera2015/usbserial #24: PlatformException(UsbSerialPortAdapter)

venil7, 2019-11-02:
I'm getting `E/flutter (29253): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(UsbSerialPortAdapter, Not an Serial device., null)`

USB device:

```
    +-o ACR1252 Dual Reader@14200000  <class AppleUSBDevice, id 0x100016c10, registered, matched, active, busy 0 (6 ms), retain 19>
```

MinhCuongIT, 2019-12-28:
Same to you. But I don't know the way to fix it.

altera2015, 2020-03-07:
Were you running this on IOS?

MinhCuongIT, 2020-03-09:
I run it on Android  API 25

ConProgramming, 2020-03-18:
I also have this... is this package abandoned, or?

altera2015, 2020-03-18:
Strange. I pushed an update last week. Are you sure you have the latest?

altera2015, 2020-03-18:
Not every serial adapter is supported, can you tell me exactly what hardware it is. Include the vid and pid.

ConProgramming, 2020-03-18:
I'm trying to connect a DigitalPersona U.are.U 4500 Fingerprint Reader
VID: 1466
PID: 10

ConProgramming, 2020-03-18:
Calling `UsbSerialDevice.isSupported(device)` on the device returns `false`, so it looks your right... thanks for the quick reply though!

altera2015, 2020-03-18:
Yeah looking at the underlying library it is not included https://github.com/felHR85/UsbSerial

closed 2021-07-11 by altera2015