altera2015/usbserial #25: Build failure after Flutter 1.12 upgrade

venil7, 2019-12-12:
I get the following error
```
FAILURE: Build failed with an exception.                                
                                                                        
* What went wrong:                                                      
Execution failed for task ':usb_serial:verifyReleaseResources'.         
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed                                    
     /Users/user/.gradle/caches/transforms-2/files-2.1/0a271e99b6771ad4a84318244d532fb7/core-1.0.0/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/fontVariationSettings not found.
                                                                        
     /Users/user/.gradle/caches/transforms-2/files-2.1/0a271e99b6771ad4a84318244d532fb7/core-1.0.0/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/ttcIndex not found.
```

JAICHANGPARK, 2019-12-19:
I have same problem

JAICHANGPARK, 2019-12-19:
Solved 
Change compileSdkVersion & targetSdkVersion   27 to 28 

```
compileSdkVersion 28

    lintOptions {
        disable 'InvalidPackage'
    }

    defaultConfig {
        // TODO: Specify your own unique Application ID (https://developer.android.com/studio/build/application-id.html).
        applicationId "dev.bessems.usbserialexample"
        minSdkVersion 24
        targetSdkVersion 28
        versionCode flutterVersionCode.toInteger()
        versionName flutterVersionName
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
    }

```

venil7, 2019-12-19:
I already had those changes, and it doesnt solve the issue:

my full build output:
```
FAILURE: Build failed with an exception.                                
                                                                        
* What went wrong:                                                      
Execution failed for task ':usb_serial:verifyReleaseResources'.         
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed                                    
     /Users/darkruby/.gradle/caches/transforms-2/files-2.1/0a271e99b6771ad4a84318244d532fb7/core-1.0.0/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/fontVariationSettings not found.
     /Users/darkruby/.gradle/caches/transforms-2/files-2.1/0a271e99b6771ad4a84318244d532fb7/core-1.0.0/res/values/values.xml:57:5-88:25: AAPT: error: resource android:attr/ttcIndex not found.
* Try:                                                                  
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
* Get more help at https://help.gradle.org                                                                     
BUILD FAILED in 58s                                                     
Running Gradle task 'assembleRelease'...                                
Running Gradle task 'assembleRelease'... Done                      59.2s
The built failed likely due to AndroidX incompatibilities in a plugin. The tool is about to try using Jetfier to solve the incompatibility.
Building plugin usb_serial...
Running Gradle task 'assembleAarRelease'...                             
Running Gradle task 'assembleAarRelease'... Done                    0.1s
env: sh\r: No such file or directory
The plugin usb_serial could not be built due to the issue above.
```

JAICHANGPARK, 2019-12-19:
@venil7 Check this error 

> The` built failed likely due to AndroidX incompatibilities in a plugin. The tool is about to try using Jetfier to solve the incompatibility.

HirenIos, 2019-12-24:
I also get same error 
```

FAILURE: Build failed with an exception.                                
                                                                        
* What went wrong:                                                      
Execution failed for task ':usb_serial:verifyReleaseResources'.         
> java.util.concurrent.ExecutionException: com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource linking failed
  /Users/admin/git/build/usb_serial/intermediates/res/merged/release/values/values.xml:236: error: resource android:attr/fontVariationSettings not found.
  /Users/admin/git/build/usb_serial/intermediates/res/merged/release/values/values.xml:237: error: resource android:attr/ttcIndex not found.
  error: failed linking references.                                     
                                                                        
                                                                        
* Try:                                                                  
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
                                                                        
* Get more help at https://help.gradle.org                              
                                                                        
BUILD FAILED in 1m 47s     
```

ManuelSIlvaCav, 2020-01-02:
This worked for me.
In the build gradle (not of your app but the one in the package) change the compileSdkVersion (from 27 to 28). So... my guess is that in the next update of this package, if they upgrade the gradle, it will solve this issue

diegomxc, 2020-01-02:
> Solved
> Change compileSdkVersion & targetSdkVersion 27 to 28
>
> ```
> compileSdkVersion 28
>
>     lintOptions {
>         disable 'InvalidPackage'
>     }
>
>     defaultConfig {
>         // TODO: Specify your own [...]

Please consider this change in the next version.

igorantolic, 2020-01-14:
I still have the same problem. I tried to create a new project, and copy the code from the old one. 
In a new project, there is already "compileSdkVersion 28"  also "targetSdkVersion 28"

Changing to 29 doesn't help.

Any suggestions?

```
Execution failed for task ':usb_serial:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed
     C:\Users\Igor\.gradle\caches\transforms-2\files-2.1\0a271e99b6771ad4a84318244d532fb7\core-1.0.0\res\values\values.xml:57:5-88:25: AAPT: error: resource android:attr/fontVariationSettings not found.

     C:\Users\Igor\.gradle\caches\transforms-2\files-2.1\0a271e99b6771ad4a84318244d532fb7\core-1.0.0\res\values\values.xml:57:5-88:25: AAPT: error: resource android:attr/ttcIndex not found.

```

diegomxc, 2020-01-14:
> I still have the same problem. I tried to create a new project, and copy the code from the old one.
> In a new project, there is already "compileSdkVersion 28" also "targetSdkVersion 28"
>
> Changing to [...]
Wich gradle are u changing ? , U need to open gradle from usbserial, not app gradle.

mrincognito39, 2020-02-20:
Yep diegomxc is correct, you will need to open the external Libraries/Flutter Plugins/usb_serial-0.2.3/android/build.gradle file and change the compile sdk version from 27 to 28.

Matt1700, 2020-03-02:
I've created a pull request that should solve this problem (it literally just changes the target SDK version to 28). Meanwhile, as @mrincognito39 said you can just change manually your pub cache. You'll find the pub cache in the Flutter SDK path (you can run `flutter pub get -v` to see the full path of the cache of your dependencies).

altera2015, 2020-03-07:
Pull Request merged. Preparing new package.

closed 2020-03-07 by altera2015