altera2015/usbserial #37: [Question] Needs some help receiving serial message without the package breaking a single message into multiple messages.

Drkstr, 2020-12-31:
Hi, 
        First of, thank you very much for this package. I have used it on multiple projects now and it has worked super well :D 

I am trying to receive and parse MAVLink message. This is the message format https://mavlink.io/en/guide/serialization.html 

I know we can do something like this to set terminator bytes
`Transaction.terminated(port.inputStream, Uint8List.fromList([48, 49]));` 
but MAVLink doesn't seem to have any terminator bytes. 

Do you have any suggestions on how I can get the package to stop breaking a single message into multiple messages?

muneebmalik78, 2021-02-24:
Hi,

I have the same problem. Does anyone have the solution for this ?? @Drkstr did you find anything relevant?

Drkstr, 2021-02-24:
Yup I ended up using the Magic Header + Length byte option described here https://github.com/altera2015/usbserial#usage-of-transaction-api

muneebmalik78, 2021-02-24:
Hi @Drkstr,

Thank you for your prompt reply. So Below is my code.

`_transaction = Transaction.magicHeader(
      _port.inputStream,
      Uint8List.fromList([239, 1, 255, 255, 255, 255]),
    );

    _transaction.stream.listen((event) {
      print(event);
    });`
 
This is frame which I send to the MCU

> 239, 1, 255, 255, 255, 255, 1, 0, 7, 13, 0, 0, 0, 0, 0, 27

And Response should be as follows

> 239, 1, 255, 255, 255, 255, 7, 0, 3, 0, 0, 10

In above code I am unable to find the length parameter. Would be great if you could provide me the example code for it or any clue would help to give the length for splitting the response.

Drkstr, 2021-03-01:
The MCU should have some documentation on the format of the response. 

Do you know if the MCU uses a framework or OS?

muneebmalik78, 2021-03-01:
Thank you @Drkstr for your response. I have done it by editing the following code snippet.

```
_transactionTemp = Transaction.magicHeader(
      _port.inputStream,
      [null, 0],
    );
```

Here it will automatically looks for the Magic header and length as well.

closed 2021-07-11 by altera2015

xty19, 2022-02-04:
> Thank you @Drkstr for your response. I have done it by editing the following code snippet.
> 
> ```
> _transactionTemp = Transaction.magicHeader(
>       _port.inputStream,
>       [null, 0],
>     );
> ```
> 
> Here it will automatically looks for the Magic header and length as well.

I refer your code, but the ide informs me that
"The element type 'Null' can't be assigned to the list type 'int'. (Documentation)"
now the null safety is update,how can we solute this problem.