altera2015/usbserial #38: Need to get USB status

realrk69, 2021-01-06:
Hi,

I'm trying to create a stream for listening to the USB status (permission allowed/denied, OTG available or not, etc.). Is there any functionality built into this library to check for the same?

here is the sample pseudo code:

```
class USBStatusMonitor implements ReactiveState<USBStatus> {
  const USBStatusMonitor(this._usb);

  final UsbSerial _usb;

  @override
  Stream<USBStatus> get state => _usb.statusStream;
}
```
Is this functionality like statusStream or USBStatus available in this library for easy detection of sudden changes from the user (user suddenly disables OTG, OTG is not available on this device etc.)?

realrk69, 2021-01-10:
I couldn't find the functionality in the API reference.

EParisot, 2021-01-21:
Hi, maybe you can just listen to the USB events and get a status from there :

`UsbSerial.usbEventStream.listen((UsbEvent msg) async {
      if (msg.event == UsbEvent.ACTION_USB_ATTACHED) {
        // Do something if usb attached
      } else if (msg.event == UsbEvent.ACTION_USB_DETACHED) {
        // Do something else if usb detached
      }
    });`
    
    If you declared your usb device(s) in android manifest, you will receive events for them...

realrk69, 2021-01-30:
Thanks, this is kind of what I was looking for. What about checking permissions or getting permissions in case the user has denied it? Is there any library which exposed that API?

EParisot, 2021-01-30:
Great !
Yes I use https://pub.dev/packages/permission_handler to ask the user about permissions...

realrk69, 2021-01-31:
Thanks m8, this works.

altera2015, 2023-09-04:
Closing since fixed.

closed 2023-09-04 by altera2015