altera2015/usbserial #41: Is there any way to read and write data in utf-8 format?

realrk69, 2021-01-31:
I'm developing a project for IoT for home sensors. I want to retrieve data stored from the devices (Atmega running on arduino bootloader).  But all the data is in string format and the device also accepts strings in the serial console. Is it possible to read and write strings using this library? Or would I have to encode and decode the messages on both ends?

realrk69, 2021-02-01:
Got it. Convert to codeunits

closed 2021-02-01 by realrk69

pattystars, 2022-03-08:
Hi Realrk95,

I am attempting to establish communication with a device that only takes UTF8 as well, but am having issues.
I am able to connect to the device, but no data is sent or received. 
I have confirmed that communication does work on my device using another app from the play store. (Serial USB Terminal)
I confirmed the serial device only accepts UTF-8 and ASCII.  When switching the serial terminal to a different charset such as UTF16 or UTF32 there is no response. 

I am attempting to send "A \r\n", but I believe that the Unit 8 list is changing the message.  
For instance A \r\n when changed with codeunits is  [65, 13, 10], and I think this is causing there to be no response. 

I tried several options such as UTF8Encode, AsciiEncoder, and hex.encode, but am still not able to get any response.  Maybe Stream<Uint8List> must be changed as well.

Any suggestions or example on how you completed this?  If the only items that can be sent with port.write are Uint8 lists wont the output always be changed?

realrk95, 2022-03-08:
> Hi Realrk95,
>
>
>
> I am attempting to establish communication with a device that only takes UTF8 as well, but am having issues.
>
> I am able to connect to the device, but no data is sent or received.
>
> I [...]

Don’t forget to set baud rate and also transaction (sending a command while awaiting a response with timeout) in case you haven’t already or are specifying an end of line character/word.

‘
await _port!.setPortParameters(
        115200, UsbPort.DATABITS_8, UsbPort.STOPBITS_1, UsbPort.PARITY_NONE);

    _transaction = Transaction.stringTerminated(
        _amsport!.inputStream!, Uint8List.fromList('OK'.codeUnits));
‘

Then:

'
await _port!
          .write(Uint8List.fromList(('A' + '\r\n').codeUnits));
'

This is the correct way to do it. If you're listening for a reply:

‘
final response = await _transaction!.transaction(
            _port!,
            Uint8List.fromList(('Send' + '\r\n').codeUnits),
            Duration(milliseconds: 100));
‘

pattystars, 2022-03-08:
Thank you for such a quick response.  I am receiving null for my response.  The port settings are confirmed to be the same as the serial port app I used.  Only settings different would be the charset and maybe the "newline" which I have set to "CR+LF".  

I believe it may have to do with the codeUnits as it converts the string to a unmodifiable list of UTF-16 code units.  The usb device only accepts UTF8 or ASCII and when sending with codeunits it is converted into [65, 13, 10].

On the testing serial app I switched the display mode to Hex and I have also found success with sending Hexidecimals 41 or 61 which is "a" and "A".  When trying to send through Uint8 list with just 41 or 61 in the list I am still getting null. 

I am wondering if I may need to use something that fixes the bytes the same length for the expected value when using Uint8list or if there is a char set that is required.  Any other suggestions you might think will work?