altera2015/usbserial #43: ch340 no detected!

Eduardom13, 2021-03-11:
Hello, I'm trying to conduct communication with the CH340 driver who owns the Nodemcu or the Arduino Nano but it does not go: c I tested with ESP32 and with this if it detects me without problems.

closed 2021-03-11 by Eduardom13

bilal3ilogic, 2021-03-24:
did you solved this issue? why did you closed it ? 
can you please tell me how if you solved it? thanks :)

Eduardom13, 2021-03-24:
delete this line from example code       
 subtitle: Text(device.manufacturerName),
now works fine, i tested in esp32, nodemcu, wemos d1 mini, arduino nano

Eduardom13, 2021-03-24:
> did you solved this issue? why did you closed it ?
> I was not able to detect ch340.
> can you please tell me how if you solved it? thanks :)

delete this line from example code       
 subtitle: Text(device.manufacturerName),
now works fine, i tested in esp32, nodemcu, wemos d1 mini, arduino nano

Zopenzop, 2022-04-16:
@Eduardom13 Did you create and use a device_filter.xml and the optional intents?