altera2015/usbserial #49: Always ask permission to use USB port

dariocavada, 2021-07-02:
Every time I open the program it asks me for authorization to use the USB port every time. 

Is it possible to remove the dialog once permissions have been granted? 
Or somehow add the permissions so that the dialog doesn't open?

altera2015, 2021-07-05:
Not sure what is going on this was certainly not the behavior intended ( not observed ). Unfortunately I don't have a USB Serial dongle at hand at the moment. Maybe someone with hardware at hand can contribute?

dariocavada, 2021-07-08:
It's an Android Problem...

I've tried all possible manifest configurations without any success.

https://android.stackexchange.com/questions/39851/how-to-stop-android-asking-for-permission-before-permitting-access-to-usb-device

https://developer.android.com/guide/topics/connectivity/usb/host

UzairJawaid, 2021-07-29:
Any solution you found?

dariocavada, 2021-07-29:
> Any solution you found?

No solution found without rooting the device...

UzairJawaid, 2021-08-05:
I am new to routing,can you please share some steps for rooting a tv box. I am using t9 tv box. https://www.daraz.pk/products/t9-4gb64gb-android-10-4k-smart-android-tv-box-i179136798-s1356836931.html?spm=a2a0e.searchlistcategory.list.1.5d10314bF377rJ&search=1

MiladEsp, 2021-10-05:
on a Samsung S8 with Android 9, I don't have this issue. I've just added an intent filter as per the repo README. Once I give the permission after connecting my device, it doesn't pop up anymore.

`<intent-filter>
      <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
</intent-filter>
`
`
<meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" android:resource="@xml/device_filter" />
`

oscarfv, 2021-11-22:
What worked for me is to create the intent as explained above, install the app with the device disconnected and connect the device. Android will ask if you want to open the application when the device is connected.

If you grant the permission when the application is already running and trying to access the device, it only will work until the next reboot.

Today a user reported that one tablet is asking for permission after every reboot, it does not ask when the device is connected. I plan to remove the permission, reboot and follow the procedure.

altera2015, 2023-09-04:
Closing since this is an android issue.

closed 2023-09-04 by altera2015