altera2015/usbserial #51: Any plans to update the library to android embedding v2?

MiladEsp, 2021-07-10:
Getting `warning: [deprecation] Registrar in PluginRegistry has been deprecated` with a clean build. Looking in the code, the reason seems to be that the library is using Android embedding v1.

Thanks

altera2015, 2021-07-11:
Interesting. Looks like this details the upgrade steps:

https://flutter.dev/docs/development/packages-and-plugins/plugin-api-migration

altera2015, 2021-07-11:
Hey MiladEsp,

Please test branch https://github.com/altera2015/usbserial/tree/EmbeddingV2 

I don't have a UsbSerial device at hand so I'm limited in what I can test at the moment. Let me know if this look good.

MiladEsp, 2021-07-11:
Thanks for the prompt response, @altera2015 .

With a clean build, I don't get those warnings anymore. Tested the example included with the library (with minor modifications specific to my device), and it's working as expected. The tested device uses the CH340 chip and the test was done on a Samsung S8.

Only thing worth mentioning, which I'm not sure if is related to your changes here, is that when I disconnect my device, I get an InterruptedException.

[image]

altera2015, 2021-07-11:
Great, glad that works. I'll do a new release soon. 

As far as that exception trace, does it ever hit any code in my plugin? 

If so I can catch it otherwise, if it doesn't kill the app, I think we might let that one go.

closed 2021-07-12 by altera2015