altera2015/usbserial #66: Exception: PlatformException(UsbSerialPortAdapter, Not an Serial device., null)

vonam2991, 2022-05-04:
I'm getting Unhandled Exception: PlatformException(UsbSerialPortAdapter, Not an Serial device., null)
when create 1 UsbDevice Prolific PL2303  <usb-device vendor-id="067B" product-id="23F3" />

rohanddave, 2023-03-31:
Facing the same problem, did you find a fix? @vonam2991

gmkado, 2023-08-21:
Same issue, device vid=6790, pid=29987 (CH340).  

I was able to get the example code from this repo to work fine with this device, but copying and pasting the `main.dart` into a new flutter project (using `flutter create` with flutter version 3.13.0) has the same issue.

I think this points to an issue with the android specific code/configuration, but haven't tracked it down further.

gmkado, 2023-08-21:
Hmm as hinted at in #70 and a bunch of other issues in this repo, changing from targetSdkVersion 33 to 30 fixed this for me.

Maintainers of this lib should consider closing some of these as duplicates...

gmkado, 2023-08-21:
(Better yet https://github.com/altera2015/usbserial/issues/75#issuecomment-1605306814)

altera2015, 2023-09-04:
Fix Merged.

closed 2023-09-04 by altera2015