altera2015/usbserial #70: Permission issues

Aagam9833, 2022-09-15:
I have used felHR85/UsbSerial library in my java android project. 

In that library there were functions of UsbService.this.serialPort.open(), which i suppose is the equivalent of UsbPort.open() function in this library. 

My usb device is getting detected perfectly with all details of manufacturer, pid, vid. However whenver I try to connect to the device it shows following error

`E/UsbManager( 4561): exception in UsbManager.openDevice
E/UsbManager( 4561): java.lang.SecurityException: User has not given 11180/com.example.curveusbserial permission to access device /dev/bus/usb/001/002`

Why am i getting this error? The felHR85/UsbSerial properly works by opening the connection to the device without throwing permission error but here it does?

RafaelSchridi, 2022-10-10:
Having the same problem currently, did you find a solution?

Aagam9833, 2022-10-11:
Nope nothing yet

RafaelSchridi, 2022-10-11:
I think i fixed this by adding 
```bash
	<intent-filter>
		<action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
	</intent-filter>

	<meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
		android:resource="@xml/device_filter" />
```
To the AndroidManifest
and addeding

corteXnzt, 2023-03-07:
> I think i fixed this by adding
> 
> ```shell
> 	<intent-filter>
> 		<action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
> 	</intent-filter>
> 
> 	<meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
> 		android:resource="@xml/device_filter" />
> ```
> 
> To the AndroidManifest and making a xml/device_filter.xml
> 
> ```
> <?xml version="1.0" encoding="utf-8"?>
> 
> <resources>
>     <usb-device vendor-id="1234"  />
> </resources>
> ```
> 
> The README says this is optional but I think it's required for recent Android versions.

After this the example is still not working with android 12. 

What I did to make it work.

I only use the intent filter part. Then on build.gradle I set this. I lowered the target sdk version from 31 to 30.
>compileSdkVersion 31
minSdkVersion 19
 targetSdkVersion 30

closed 2023-09-04 by altera2015