altera2015/usbserial #75: USB Permission Ask Problem

ImperialCrise, 2023-02-09:
In the old android with an android < 7.0, this library works fine. But with my android 12 i got an excepetion when i try to create an port with device.create().

The exceeption i got is : 

PlatformExcepetion(UsbSerialPortAdapter, Failed to acquire USB device., null, null).

It seems like that issue https://bytemeta.vip/index.php/repo/woodemi/quick_usb/issues/60 but for another library.

rohanddave, 2023-03-31:
Try setting `targetSdkVersion 30` in your app `build.gradle` @ImperialCrise

magnus-lpa, 2023-06-24:
There is an already merged pull request (c20e1e6e171137db4b52413dac14333653f07905) that fixes this, but for some reason the plugin itself has not been updated. You can use the relevant commit from Github like this in your pubspec.yaml (i.e. instead of `usb_serial: ^0.4.0`):

  usb_serial:
    git:
      url: https://github.com/altera2015/usbserial.git
      ref: c20e1e6e171137db4b52413dac14333653f07905

closed 2023-09-04 by altera2015