altera2015/usbserial #92: can I use OTG?

aldysetiaa, 2023-09-15:
I want to use OTG to USB -> RS232 ATEN, and serial to the scale indicator, is that possible?

rockerer, 2023-12-12:
Hi,
i'm using this library with an OTG [adapter](https://www.amazon.com/MOSWAG-Splitter-Kompatibel-Chromecast-Samsung-schwarz/dp/B08Y6VQN99?ref_=ast_sto_dp) from Moswag and a Samsung Active and have no issues. So the best thing might be to just try it. It looks like the [adapter](https://www.amazon.com/Aten-UC232A-USB-Serial-Converter/dp/B000J3OK7C?th=1)  has a USB-A connector, so the question arises how you want to connect this adapter to your android device. 

If you have further question regarding this issue, you can just respond in this thread.

rockerer, 2024-04-02:
Hi @aldysetiaa,
did you make any progress regarding this issue or do you have any further questions?
Greetings