altera2015/usbserial #93: Error in the example

paulobreim, 2023-10-01:
I am trying to run the example showed in the lib and I am getting the follow error.

[image]

What is wrong?

tks

rockerer, 2023-12-12:
Hi,
what platform are you running the code on? The plugin is written for android, so running it on ios, linux or windows might result in the error shown above.

paulobreim, 2023-12-12:
I tested with linux and android.
I didn't notice it was only for Android.

Is there any prospect of it working on Windows?

tks

rockerer, 2023-12-13:
AFAIK it is not planned, but @altera2015 might say more about the plans regarding his package.

You might be interested in using one of the following packages, which support serial communication for linux and/or windows:
- [flutter_libserialport](https://pub.dev/packages/flutter_libserialport)
- [libserialport](https://pub.dev/packages/libserialport)

If there are more people interested in extending this package to support windows / linux / ios it might be possible to do so.

paulobreim, 2023-12-13:
TKS,  I will check this libs.

I want to make a application to share USB over the internet. I saw some apps in the internet, but nor open.

rockerer, 2023-12-14:
Do you need a graphical userinterface for this purpose or why are you using flutter?

I think, it's easier to use a language like python, c++ or Golang to achieve this goal, if you don't need the UI. These languages might have an even better (or even native) support for the serial connection and support different approaches to share the connection over the internet.

rockerer, 2024-04-02:
Hi @paulobreim,
we discussed about the issue and i think there is no reason to keep this issue open anymore. Would you mind closing this issue.
Thanks

closed 2024-07-12 by altera2015