altera2015/usbserial #97: Byte empty / errore

Taverz, 2023-11-21:
E/flutter (17985): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: SerialPortError: Interrupted system call, errno = 4
E/flutter (17985):
I/flutter (17985): port.inputStream!.listen[0]
I/FTDISerialDevice(17985): Control Transfer Response: 0
I/FTDISerialDevice(17985): Control Transfer Response: 0
W/System.err(17985): java.lang.InterruptedException
W/System.err(17985): 	at java.lang.Object.wait(Native Method)
W/System.err(17985): 	at java.lang.Object.wait(Object.java:442)
W/System.err(17985): 	at java.lang.Object.wait(Object.java:568)
W/System.err(17985): 	at com.felhr.usbserial.SerialBuffer$SynchronizedBuffer.get(SerialBuffer.java:117)
W/System.err(17985): 	at com.felhr.usbserial.SerialBuffer.getWriteBuffer(SerialBuffer.java:72)
W/System.err(17985): 	at com.felhr.usbserial.UsbSerialDevice$WriteThread.doRun(UsbSerialDevice.java:402)
W/System.err(17985): 	at com.felhr.usbserial.AbstractWorkerThread.run(AbstractWorkerThread.java:21)
I/flutter (17985): port.inputStream!.listen[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
I/FTDISerialDevice(17985): Control Transfer Response: 0
I/FTDISerialDevice(17985): Control Transfer Response: 0
Application finished.

rockerer, 2024-04-10:
Can you please give some context to this message. When does it happen, what devices are used?

altera2015, 2024-07-12:
close for lack of reply.

closed 2024-07-12 by altera2015