altera2015/usbserial #98: Readme unclear compileOptions instructions

guyluz11, 2023-12-24:
Readme states that it is important to add `compileOptions` into build.gradle  but the example link does not contain `compileOptions` 

Readme
https://github.com/altera2015/usbserial#important-appbuildgradle

Example link
https://raw.githubusercontent.com/altera2015/usbserial/master/example/android/app/build.gradle

guyluz11, 2023-12-24:
I see from the git history that it got removed, so we need to remove it also from redme

rockerer, 2024-02-05:
Thanks for pointing out the mistake. Would you like to create a pull request to fix this, if you already found out, what's wrong.

guyluz11, 2024-02-05:
Sorry, I am currently working on other things.
Maybe next time while I am using the package.