altera2015/usbserial #99: Write files in pendrive

Bestfastfire, 2023-12-31:
Is it possible to use this lib to save files to a pendrive?

I tried doing this:
```
    final bytes = <int>[];

    await yt.videos.streamsClient.get(audioStream).listen((chunk) {
      bytesReceived += chunk.length;
      bytes.addAll(chunk);

    }, onDone: () async{
      MyLog.log('Done onDone');

    }, onError: (error) async{
      MyLog.log('Done error: $error');

    }).asFuture();

    final devices = await UsbSerial.listDevices();
    final port = await devices.first.create();
    final openResult = await port?.open();

    if ( !(openResult ?? false)) {
      MyLog.log("Failed to open");
      return;

    }

    port?.inputStream?.listen((Uint8List event) {
      MyLog.log('Event close');
      port.close();
    });

    await port?.write(Uint8List.fromList(bytes));
```

But I get the error:

`[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(UsbSerialPortAdapter, Not an Serial device., null, null)`

rockerer, 2024-02-05:
Hi,
what exactly are you trying to achieve? What system are you working on (android / ios / web / linux / ...)

If the device you want to write to is mounted, the package [path_provider](https://pub.dev/packages/path_provider) might be a better choice.

If you really want to use this package, you would have to handle all the low level stuff, which is normally done by the operating system.

rockerer, 2024-02-06:
To give your users the possibility to choose the location [file_picker](https://pub.dev/packages/file_picker) is another idea.

Bestfastfire, 2024-02-06:
> To give your users the possibility to choose the location [file_picker](https://pub.dev/packages/file_picker) is another idea.

It's not working either lol, I explained it better on stack overflow, but in file_picker, when you select the path inside the pendrive it only returns "/"

altera2015, 2024-07-12:
This doesn't sound like a supported serial device.

closed 2024-07-12 by altera2015