altera2015/usbserial #103: Example app transmits data but does not receive data on Android

jnorre, 2024-02-21:
Tested with connection from PC to Android and from embedded device to Android: Example app connects and transmit fine, but does not receive data - the _subscription function (line 68ff) is never called.
Phone: Poco X3 NFC, Android 12
USB dongles: FTDI USB-RS232-3.3 and Prolific PL2303
Both dongles tested  and working correctly with "Serial USB Terminal" app from Play store.

rockerer, 2024-02-28:
Hi,
i just tested the example App from this repo with a Moswag OTG-Adapter and a serial device (embedded device using a ESP32 via FTDI for serial) and it works perfectly fine. The mentioned line (70: `_subscription = ...`) is called when connecting the tablet to the adapter. So i doubt this is an issue with the example itself.

Are there any debug informations available? What happens, when stepping through the code line by line? Maybe the connection is not established correctly?

Greetings

jnorre, 2024-03-04:
Hi,
Thank for your feedback. I tried the example on anotehr computer (linux) with a fresh install of flutter and it worked as expected. Then I retried on the original computer (Windows) and it also worked as expected - only change was an updated version of dart. So you are right - the example works perfectly!

closed 2024-03-04 by jnorre

rockerer, 2024-03-06:
Nice to hear, that the problem could be solved and thanks for closing the issue.