altera2015/usbserial #104: java.lang.InterruptedException on close invocation

camillobucciarelli, 2024-03-07:
When I try to close connection with `port.close();` I get the following error in console:

```
D/UsbDeviceConnectionJNI(28094): close
W/System.err(28094): java.lang.InterruptedException
W/System.err(28094): 	at java.lang.Object.wait(Native Method)
W/System.err(28094): 	at java.lang.Object.wait(Object.java:386)
W/System.err(28094): 	at java.lang.Object.wait(Object.java:524)
W/System.err(28094): 	at com.felhr.usbserial.SerialBuffer$SynchronizedBuffer.get(SerialBuffer.java:117)
W/System.err(28094): 	at com.felhr.usbserial.SerialBuffer.getWriteBuffer(SerialBuffer.java:72)
W/System.err(28094): 	at com.felhr.usbserial.UsbSerialDevice$WriteThread.doRun(UsbSerialDevice.java:402)
W/System.err(28094): 	at com.felhr.usbserial.AbstractWorkerThread.run(AbstractWorkerThread.java:21)
```

there is something that I'm doing wrong?

Thank you

rockerer, 2024-04-02:
Hi,
can you provde a small code example, how you use the lib? Some information about the used hardware is helpful, too. 
Thanks

rockerer, 2024-06-26:
Does your android device work with the CH34xSerialDevice when using the example app? What does the error look like?

zyq9613, 2025-02-17:
`dart `
Future<String> _readResponseWithTimeout(UsbPort port) async {
    print('1@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@');
    final completer = Completer<String>();
    port.inputStream?.listen((Uint8List data) {
      if (data.isEmpty) {
        if (!completer.isCompleted) {
          completer.complete("");
        }
      }
      print('received: $data');
      String hexString =
          data.map((byte) => byte.toRadixString(16).padLeft(2, '0')).join();

      if (!completer.isCompleted) {
        completer.complete(hexString.toUpperCase());
      }
    });
    // 设置超时
    Future.delayed(const Duration(milliseconds: timeoutMs), () async {
      if (!completer.isCompleted) {
        completer.complete("");
      }
    });
    return completer.future;
  }

  Uint8List hexStringToUint8List(String hexString) {
    // 去除空格并将字符串转换为大写
    hexString = hexString.replaceAll(' ', '').toUpperCase();
    // 创建一个空的列表来存储转换后的字节
    List<int> byteList = [];
    // 遍历每两个字符，将其转换为十六进制字节
    for (int i = 0; i < hexString.length; i += 2) {
      String hexByte = hexString.substring(i, i + 2); // 取出每两位字符
      byteList.add(int.parse(hexByte, radix: 16)); // 将两位字符转换为十六进制字节
    }
    print(Uint8List.fromList(byteList));
    // 返回对应的 Uint8List
    return Uint8List.fromList(byteList);
  }

  void initialize({String? portName}) async {
    if (_serialPort != null) return;
    UsbPort? port;
    try {
      List<UsbDevice> devices = await UsbSerial.listDevices();
      if (devices.isEmpty) {
        print("没有可用的USB设备");
        return;
      }
      Uint8List byteArray = hexStringToUint8List(hexString);
      for (UsbDevice device in devices) {
        // 创建端口
        port = await device.create();
        bool? openResult = await port?.open();
        if (openResult == null || !openResult) {
          print("打开串口失败: ${device.toString()}");
          continue;
        }
        print('当前的串口: ${port.toString()}');
        await port!.setDTR(true);
        await port.setRTS(true);
        port.setPortParameters(115200, UsbPort.DATABITS_8, UsbPort.STOPBITS_1,
            UsbPort.PARITY_NONE);
        port.setFlowControl(UsbPort.FLOW_CONTROL_OFF);
        // 发送消息
        await port.write(byteArray);
        // 获取返回信息
        String data = await _readResponseWithTimeout(port);
        print("Received data: $data");
        // 当前关闭的时候会报错，等官方修复，不影响其他操作
        await port.close();
        if (data.contains('FE')) {
          _serialPort = port;
          return;
        }
      }
    } catch (e) {
      print("初始化失败: $e");
    } finally {}
  }

zyq9613, 2025-02-17:
`Future<String> _readResponseWithTimeout(UsbPort port) async {
    print('1@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@');
    final completer = Completer<String>();
    port.inputStream?.listen((Uint8List data) {
      if (data.isEmpty) {
        if (!completer.isCompleted) {
          completer.complete("");
        }
      }
      print('received: $data');
      String hexString =
          data.map((byte) => byte.toRadixString(16).padLeft(2, '0')).join();

      if (!completer.isCompleted) {
        completer.complete(hexString.toUpperCase());
      }
    });
    // 设置超时
    Future.delayed(const Duration(milliseconds: timeoutMs), () async {
      if (!completer.isCompleted) {
        completer.complete("");
      }
    });
    return completer.future;
  }

  Uint8List hexStringToUint8List(String hexString) {
    // 去除空格并将字符串转换为大写
    hexString = hexString.replaceAll(' ', '').toUpperCase();
    // 创建一个空的列表来存储转换后的字节
    List<int> byteList = [];
    // 遍历每两个字符，将其转换为十六进制字节
    for (int i = 0; i < hexString.length; i += 2) {
      String hexByte = hexString.substring(i, i + 2); // 取出每两位字符
      byteList.add(int.parse(hexByte, radix: 16)); // 将两位字符转换为十六进制字节
    }
    print(Uint8List.fromList(byteList));
    // 返回对应的 Uint8List
    return Uint8List.fromList(byteList);
  }

  void initialize({String? portName}) async {
    if (_serialPort != null) return;
    UsbPort? port;
    try {
      List<UsbDevice> devices = await UsbSerial.listDevices();
      if (devices.isEmpty) {
        print("没有可用的USB设备");
        return;
      }
      Uint8List byteArray = hexStringToUint8List(hexString);
      for (UsbDevice device in devices) {
        // 创建端口
        port = await device.create();
        bool? openResult = await port?.open();
        if (openResult == null || !openResult) {
          print("打开串口失败: ${device.toString()}");
          continue;
        }
        print('当前的串口: ${port.toString()}');
        await port!.setDTR(true);
        await port.setRTS(true);
        port.setPortParameters(115200, UsbPort.DATABITS_8, UsbPort.STOPBITS_1,
            UsbPort.PARITY_NONE);
        port.setFlowControl(UsbPort.FLOW_CONTROL_OFF);
        // 发送消息
        await port.write(byteArray);
        // 获取返回信息
        String data = await _readResponseWithTimeout(port);
        print("Received data: $data");
        // 当前关闭的时候会报错，等官方修复，不影响其他操作
        await port.close();
        if (data.contains('FE')) {
          _serialPort = port;
          return;
        }
      }
    } catch (e) {
      print("初始化失败: $e");
    } finally {}
  }`