altera2015/usbserial #105: Break

HernandoSergio, 2024-03-14:
I need to send a break through the serial line to awaken the device (i.e. lower the TX line for more than one byte time). Is it possible to do it somehow?

tommie, 2024-05-06:
No, unfortunately not. And it's not even implemented in the underlying Android library: https://github.com/felHR85/UsbSerial/blob/7ad6c9f6880e93a29498e479d3e871d5b03d188b/usbserial/src/main/java/com/felhr/usbserial/CDCSerialDevice.java#L251

closed 2024-07-12 by altera2015

Trescode-repos, 2024-10-25:
in your code:
await _port!.setBreak(false); // when you set DTR and RTS

  to usb_serial.dart line 176:
  /// Sets or clears the break port to value [breakState].
  Future<void> setBreak(bool breakState) async {
    _breakState = breakState;
    return await _channel.invokeMethod("setBreak", {"breakState": breakState});
  }
  
  
  to UsbSerialPortAdapter.java line 138:
              case "setBreak":{
                boolean v = call.argument("breakState");
                m_SerialDevice.setBreak(v);
                result.success(null);
                break;
                }

tommie, 2024-10-25:
Note that only the [FTDI](https://github.com/felHR85/UsbSerial/blob/7ad6c9f6880e93a29498e479d3e871d5b03d188b/usbserial/src/main/java/com/felhr/usbserial/FTDISerialDevice.java#L365) and [CP2102](https://github.com/felHR85/UsbSerial/blob/7ad6c9f6880e93a29498e479d3e871d5b03d188b/usbserial/src/main/java/com/felhr/usbserial/CP2102SerialDevice.java#L334) drivers have `setBreak` implemented. All the others will silently do nothing.