altera2015/usbserial #106: Open port

Guimarte, 2024-04-03:
Hi, im try using this package to get weight from a scale toledo, but i always receive false in open.port()
can u help me?

rockerer, 2024-04-10:
Hi,
can you provide more information about the following:
- used hardware
- sample code (minimal example / the important parts of your app)
- does your tablet and hardware work with the example app of this package?

I'm pretty sure, we are able to help you ;)

Guimarte, 2024-04-10:
Hi, I contacted the manufacturer of the scale and they don't have
support for Android, as it is a serial scale, to connect I have to use
a USB cable with a COM port adapter, and the device I'm using is a
tablet with inputs usb, used for POS (Point of Sale), I don't think
this way will work to get the weight coming from the scale

Message ID: ***@***.***>
>

rockerer, 2024-04-10:
Hi,
thanks for the fast answer.
It might be an idea to use an USB-OTG-Adapter to get a working serial connection. 

Tablet <---> OTG <---> USB2COM <---> Scale

Any opinions about this?

Guimarte, 2024-04-11:
in the truth my tablet have a usb port, and i use usb2com - scale

closed 2024-07-12 by altera2015