altera2015/usbserial #108: reconnect to previous device

darkmoon3d, 2024-04-10:
How would I setup within code so that app will reconnect to previous device when either app is opened again or connection is lost?

rockerer, 2024-04-10:
Can you provide more information, when you want what to happen?

You can use the IntentFilter to start the app when a device is connected. When you use the IntentFilter the app already has the permission to connect to the device.

You can write code in the init-part of your app to query the list of ports and connect to the (first) device. This connecting process will ask for permission, which has to be granted by the user.

Another idea is to start a scan every x seconds, when no device is connect or when the connection is lost and try to connect, if a device is found.

darkmoon3d, 2024-04-11:
Was thinking more of collecting "serial" of usb device and holding persistent value so when app is restarted (or if usb is detached/reattached or power cycled) the code would recognize and automatically reattach.