altera2015/usbserial #112: UsbDevice.create() does nothing instead of connecting to the serial device in Flutter 3.22

denshion, 2024-04-30:
The following minimal working example successfully connected to a CH340 device in Flutter 3.19, but stopped working in Flutter 3.22

```
import 'dart:async';
import 'dart:convert';

import 'package:flutter/material.dart';
import 'package:usb_serial/usb_serial.dart';

void main() {
  runApp(const MyApp());
}

Stream<String> getStream() {
  StreamController<String> sc = StreamController<String>();
  connect(sc);
  return sc.stream;
}

Future<void> connect(StreamSink<String> inputDataSink) async {
  inputDataSink.add('Finding devices...');
  List<UsbDevice> devices = await UsbSerial.listDevices();

  if (devices.isEmpty) {
    inputDataSink.add('Error: No UART devices');
    return;
  }
  inputDataSink.add('Found ${devices[0]}, creating a port...');
  // UsbPort? port = devices[0].port ?? await devices[0].create();
  UsbPort? port = await devices[0].create();

  if (port == null) {
    inputDataSink.add('Error: Failed to create a virtual UART port');
    return;
  }
  inputDataSink.add('Opening port $port');
  bool openResult = await port.open();
  if (!openResult) {
    inputDataSink.add('Error: Failed to open a virtuial UART port');
    return;
  }

  await port.setDTR(true);
  await port.setRTS(true);

  port.setPortParameters(
      115200, UsbPort.DATABITS_8, UsbPort.STOPBITS_1, UsbPort.PARITY_NONE);

  inputDataSink.add('Connected, here comes the data:');
  inputDataSink.addStream(port.inputStream!
    .cast<List<int>>()
    .transform(const AsciiDecoder())
    .transform(const LineSplitter())
  );
}

String data = '';

class MyApp extends StatelessWidget {

  const MyApp();

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      builder: (context, child) {
        return Scaffold(
          body: StreamBuilder(
            stream: getStream(),
            builder: (context, snapshot) {
              data += '${snapshot.data ?? ''}\n';
              return Text(data);
            }
          ),
        );
      },
    );
  }
}
```

In Flutter 3.19 it asked for a permission to connect to the device, but in Flutter 3.22 it instead hangs forever at `await devices[0].create()`, without ever asking for a permission.

**Note:** in my other large project `devices[0].create()` throws `PlatformException(UsbSerialPortAdapter, Failed to acquire USB device., null, null)` instead, but I was not able to reproduce that in the minimal example.

[This issue](https://github.com/felHR85/UsbSerial/issues/370) might be related, but I'm not sure.

## Steps to reproduce
Steps to reproduce include compiling the app using different versions of Flutter and different AndroidManifest.xml versions.
Expected behavior: the app asks for a permission to connect to the device, and after obtaining it, prints the data from the device.
### Example from this repo
1. Go to your flutter home and execute `git checkout beta` (as of April, 2024, this will download Flutter 3.22.0-0.3.pre; you may as well run `git checkout 3.22.0-0.3.pre` instead to specify the exact version)
2. Open the example from this repository
3. Replace main.dart with my MWE
Observed behavior: as expected (the app asks for a permission to connect to the device, and after obtaining it, prints the data from the device)
### Flutter 3.22
1. Create a new flutter project using Flutter 3.22 and add the github version of `usb_serial` to `pubspec.yaml`
2. Replace main.dart with my MWE
3. AndroidManifest.xml will be different, compared to its counterpart in the example project here.
Observed behavior: the app stops printing after "Found UsbSerial, creating a port...", hanging at `await devices[0].create()`, without ever asking for a permission.
### Flutter 3.19
1. Go to your flutter home and execute `git checkout stable` (as of May, 2024, this will download Flutter 3.19.6)
2. Create a new flutter project using Flutter 3.19 and add the github version of `usb_serial` to `pubspec.yaml`
3. Replace main.dart with my MWE
4. AndroidManifest.xml will be the same as in the Flutter 3.22 project.
However, the app works like expected, asking for the permission and printing the data from the device.

Asking for permissions in the manifest (#49) has no effect.

I suppose it does not really matter, but for the sake of completeness: the host is a Samsung A54 smartphone running Android 14, the device is a Lolin NodeMCU board, which is connected to the phone through an USB OTG adapter.

Foxushka, 2024-05-01:
You can retry on my branch: https://github.com/altera2015/usbserial/pull/110
Should be fixed there

denshion, 2024-05-09:
@Foxushka I confirm the issue to be fixed in your branch, let's wait for the pull request acceptance.

Once again, thank you for letting me know about your fix.