altera2015/usbserial #115: Not showing Permission Popup for targetSdk =34

Sawansb, 2024-07-12:
I was using the usb_serial: ^0.5.1 package, and it was working perfectly fine. However, after upgrading my target SDK to 34 as per Google Play policy, the usb_serial package is now giving me exceptions. Any suggestions on how to resolve this issue?

`E/UsbManager( 9781): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
E/UsbManager( 9781): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/UsbManager( 9781): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/UsbManager( 9781): 	at android.os.Handler.handleCallback(Handler.java:959)
E/UsbManager( 9781): 	at android.os.Handler.dispatchMessage(Handler.java:100)
E/UsbManager( 9781): 	at android.os.Looper.loopOnce(Looper.java:232)
E/UsbManager( 9781): 	at android.os.Looper.loop(Looper.java:317)
E/UsbManager( 9781): 	at android.app.ActivityThread.main(ActivityThread.java:8501)
E/UsbManager( 9781): 	at java.lang.reflect.Method.invoke(Native Method)
E/UsbManager( 9781): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
E/UsbManager( 9781): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
E/UsbManager( 9781): Caused by: android.os.RemoteException: Remote stack trace:
E/UsbManager( 9781): 	at com.android.server.usb.UsbUserPermissionManager.checkPermission(UsbUserPermissionManager.java:678)
E/UsbManager( 9781): 	at com.android.server.usb.UsbHostManager.openDevice(UsbHostManager.java:565)
E/UsbManager( 9781): 	at com.android.server.usb.UsbService.openDevice(UsbService.java:296)
E/UsbManager( 9781): 	at android.hardware.usb.IUsbManager$Stub.onTransact(IUsbManager.java:501)
E/UsbManager( 9781): 	at android.os.Binder.execTransactInternal(Binder.java:1496)
E/UsbManager( 9781): 
D/CompatibilityChangeReporter( 9781): Compat change id reported: 236704164; UID 11141; state: ENABLED
E/flutter ( 9781): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(UsbSerialPortAdapter, Failed to acquire USB device., null, null)
E/flutter ( 9781): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651:7)
E/flutter ( 9781): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter ( 9781): <asynchronous suspension>
E/flutter ( 9781): #2      UsbSerial.createFromDeviceId (package:usb_serial/usb_serial.dart:460:33)
E/flutter ( 9781): <asynchronous suspension>
E/flutter ( 9781): #3      UsbDevice.create (package:usb_serial/usb_serial.dart:350:13)
E/flutter ( 9781): <asynchronous suspension>
E/flutter ( 9781): #4      UsbManager.connectTo (package:serial_usb/usb_manager.dart:150:13)
E/flutter ( 9781): <asynchronous suspension>
E/flutter ( 9781): #5      _AvailableDeviceScreenState.build.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:serial_usb/views/device_available_screen.dart:179:53)
E/flutter ( 9781): <asynchronous suspension>`

Sawansb, 2024-07-12:
I have found the solution for Android 14 support. Please also update the package on pub.dev

gkres, 2024-12-01:
> I have found the solution for Android 14 support. Please also update the package on pub.dev

That's great. Would you mind sharing the solution?

vaishnavijlk, 2025-06-30:
what is the solution @Sawansb