altera2015/usbserial #118: Failed to transform UsbSerial-6.1.0.aar (com.github.felHR85:UsbSerial:6.1.0) to match attributes ..

ahmadms-dev, 2024-10-27:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Failed to transform UsbSerial-6.1.0.aar (com.github.felHR85:UsbSerial:6.1.0) to match attributes {artifactType=android-aar-metadata, org.gradle.category=library, org.gradle.libraryelements=jar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Could not find UsbSerial-6.1.0.aar (com.github.felHR85:UsbSerial:6.1.0).
        Searched in the following locations:
            https://jitpack.io/com/github/felHR85/UsbSerial/6.1.0/UsbSerial-6.1.0.aar
            https://jitpack.io/com/github/felHR85/UsbSerial/6.1.0/UsbSerial-6.1.0.jar

xwork1, 2024-11-01:
I have the same problem. When will it be solved?

` implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
artifact {
classifier = 'release'
type = 'aar'
}
}` I changed it but the problem persisted because I did not add "-release" help :((

jymden, 2024-11-01:
This is an issue directly on the UsbSerial repo as well https://github.com/felHR85/UsbSerial/issues/375

xwork1, 2024-11-01:
Oh, yeah, that worked for me. I'm going to build.gradle in the path “AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.2\android” 

“dependencies {
implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
artifact {
classifier = 'release'
type = 'aar'
}
}
}”
I added it. it worked.

> This is an issue directly on the UsbSerial repo as well [felHR85/UsbSerial#375](https://github.com/felHR85/UsbSerial/issues/375)

Oh, yeah, that worked for me. I'm going to build.gradle in the path “AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.2\android” 

```
“dependencies {
implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
artifact {
classifier = 'release'
type = 'aar'
   }
  }
}”
```
I added it. it worked.

jymden, 2024-11-02:
I forked and implemented that temp fix. So if you want you can use that repo while waiting for this to be resolved.
Just change the usb_serial dependency in pubspec.yaml  to:
```
  usb_serial:
    git:
      url: https://github.com/jymden/usbserial.git
      ref: master
```

magnus-lpa, 2025-02-26:
> Oh, yeah, that worked for me. I'm going to build.gradle in the path “AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.2\android”
>
> ```
> “dependencies { [...]

This seems to work. 

However, it doesn't use the exact same version of the underlying `UsbSerial` library, does it? The commit it points to 

https://github.com/felHR85/UsbSerial/commit/7ad6c9f6880e93a29498e479d3e871d5b03d188b

is not the same as the commit tagged with version 6.1.0

https://github.com/felHR85/UsbSerial/commit/08ef5bbb3fb2352b0dcc197564eff2d6df8678e4

So it's not exactly the same as the original implementation, right? 

Might not be a problem, but it makes me want to test the apps functionality more thoroughly if I know that the underlying serial port plugin has changed since my last build.