altera2015/usbserial #121: Looking for a maintainer

altera2015, 2024-12-01:
I've not had time to work on this in a long time nor any need for the functionality. It's probably good if we find a new maintainer. Please comment below if you are interested. People with prior accepted PR's preferred.

nadeem-agva, 2024-12-09:
Hi, myself Nadeem Akhtar.
I am willing to work for this library, having vast experience in USB and BLE communications.

altera2015, 2024-12-15:
Sorry I'd prefer someone with a github history and preferably someone who has contributed previously to this repo.

mbalmer, 2026-10-13:
As this package has issues (it does not even build on recent Flutter), I am willing to help.  I work in the POS area and have need for this.

zeyus, 2026-10-27:
I would be happy to help maintain, I'm using this package for my EEG viewer app for the lab and the public LSL version, they use `usb_serial` to communicate with an EEG amplifier and works with the OpenBCI cyton / daisy board (I've been testing that today).

Zalexei, 2026-11-02:
I hit the JitPack build error too and ended up writing a replacement:
https://pub.dev/packages/usb_serial_next

The usb-serial-for-android drivers are bundled into the plugin, so there's no
JitPack, and it works on Flutter 3.27+. So far I've only tested it with an
ESP32-S3. If you try it with CH340, CP2102, FTDI or PL2303, let me know how it goes.

Thanks for the original package!

altera2015, 2026-11-02:
@zeyus I've invited you as a collaborator, let's see how I can give you access or if we need to move the project to your github account.

zeyus, 2026-11-02:
@altera2015 Thanks. Let me know if you want to chat about the project and what level of involvement / ownership you want (and where you would want to have that discussion). I don't feel a need to rush anything.

zeyus, 2026-11-11:
@altera2015 I just released a new version. I cannot upload it to pub.dev, but if you want to you could enable "publishing to pub.dev via github actions" which would allow the version to be updated there.

(the action is here, it can be re-run once the setting is enabled and it should work, i use the same action for my other pub.dev packages) https://github.com/altera2015/usbserial/actions/runs/43015966845/job/130507422947

Thanks