altera2015/usbserial #123: Could not find com.github.felHR85:UsbSerial:6.1.0

orange-jacky, 2025-01-10:
app_flutter git:(master) ✗ fvm flutter build apk --release

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':usb_serial:generateReleaseRFile'.
> Could not resolve all files for configuration ':usb_serial:releaseCompileClasspath'.
   > Could not find com.github.felHR85:UsbSerial:6.1.0.
     Required by:
         project :usb_serial

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4m 54s
Running Gradle task 'assembleRelease'...                          295.4s
Gradle task assembleRelease failed with exit code 1

abdallah-odeh, 2025-01-11:
Same issue here

orange-jacky, 2025-01-15:
> Same issue here

  vi  pubspec.yaml 
   # usb_serial: ^0.5.2
  usb_serial:
    git:
      url: https://github.com/jymden/usbserial.git
      ref: master

flutter clean
futter run 

it‘s working